brilliantlabsAR/monocle-micropython #4: Implement media transfer over Bluetooth

josuah, 2022-11-09:


josuah, 2022-11-09:
An extra Bluetooth service provides an extra stream used for binary data.

josuah, 2023-01-06:
Waiting an app version that can pair with the Monocle.

josuah, 2023-01-09:
Could not pair insofar with two phones.
Will work on that with Umar and Raj tomorrow.

sena-mike, 2023-02-07:
Is there a spec available for the BLE services and characteristics offered over BLE?

josuah, 2023-02-08:
The main bluetooth stream giving the python console is a standard [1] Nordic Uart Service, as the upstream micropython uses.

There is a secondary stream, using the same kind of TX and RX characteristics, that is used to transfer files.
@siliconwitch did a nice doc for it at https://docs.brilliantmonocle.com/micropython/micropython/#under-the-hood

josuah, 2023-02-08:
The web REPL is also an interesting way to explore both protocols, as it gives a practical implementation to check against:
https://github.com/brilliantlabsar/web-bluetooth-repl

closed 2023-03-03 by siliconwitch