brilliantlabsAR/monocle-micropython #79: Add tests and fix bugs for the python API and modules

josuah, 2023-02-02:


josuah, 2023-02-02:
@siliconwitch 
How about we remove this test:
```
Failed - time.zone('1:20') == None. Expected: <class 'ValueError'>
```
For instance, there are time-zones with 15 minutes offset?

siliconwitch, 2023-02-02:
+1:20 is an invalid timezone. The minutes can only be :00, :30 or :45

Here's my code for reference:
```c
STATIC mp_obj_t time_zone(size_t n_args, const mp_obj_t *args)
{
    if (n_args == 0)
    {
        char timezone_string[] = "+00:00";
        sprintf(timezone_string,
                "%02d:%02u",
                time_zone_hour_offset,
                time_zone_minute_offset);

        return mp_obj_new_str(timezone_string, strlen(timezone_string));
    }

    int hour = 0;
    int minute = 0;

    if (sscanf(mp_obj_str_get_str(args[0]), "%d:%d", &hour, &minute) != 2)
    {
        mp_raise_ValueError("format provided must be '±hh:mm'");
    }

    if (hour < -12.0 || hour > 14.0)
    {
        mp_raise_ValueError("hour value must be between -12 and +14");
    }

    if (minute != 0 && minute != 30 && minute != 45)
    {
        mp_raise_ValueError("minute value must be either 00, 30, or 45");
    }

    time_zone_hour_offset = hour;
    time_zone_minute_offset = minute;

    return mp_const_none;
}
STATIC MP_DEFINE_CONST_FUN_OBJ_VAR_BETWEEN(time_zone_obj, 0, 1, time_zone);
```

josuah, 2023-02-02:
I do not think there is anything hardware-specific in the time module.

josuah, 2023-02-02:
:+1: problem solved by chat.

closed 2023-02-13 by siliconwitch