brilliantlabsAR/monocle-micropython #141: [bug] firmware crashing upon touching a button

josuah, 2023-03-03:
Simone could come-up with a firmware making the monocle reboot: go from whatever is being displayed to the default pattern.

> my crash has nothing to do with memory, at the time of it grows or about 32 kb free

Video of the crash:
https://cdn.discordapp.com/attachments/976634834879385621/1081207539187855360/VID_20230303_143013.mp4

Code leading to this crash:
```python
import display
import device
import uasyncio
import gc
import time
import touch

battlev = ""
firstaccess = True
battlevint = 100
app = ["phone", "camera" , "next?"]
idx = 0
alive = "MONOCLE"
inapp = ""
time = "x"

async def battery_level():
	global battlevint
	
	if device.battery_level() < battlevint:
		battlevint = device.battery_level()

	battlev = str(battlevint) + '%'
	display.text(battlev, 0,0,0xffffff)
	uasyncio.sleep_ms(5000)

def change_text(button):
	global app
	global idx
	global alive
	global lastapp
	global nextapp

	lastapp = alive
	limit = len(app) - 1
	
	if button == "A":
		idx = idx + 1
		if idx > limit:
			idx = 0
	if button == "B":
		idx = idx - 1
		if idx < 0:
			idx = limit
	alive = app[idx]

	draw_ui()
	persistent()
	print(alive)

def draw_ui():
	
	global idx
	uasyncio.run(battery_level())

	x = 1
	y = 100
	size = 200
	colora = 0xffffff
	colorb = 0xffffff
	colorc = 0xffffff
	space = 10

	if idx <= 2:
		if idx == 0:
			colora = 0xf8db5E
		if idx == 1:
			colorb = 0xf8db5E
		if idx == 2:
			colorc = 0xf8db5E

	# Primo quadrato
	display.hline(x, y, size, colora)
	display.vline(x, y, size, colora)
	display.vline(x + size, y, size, colora)
	display.hline(x, y + size, size, colora)

	# Secondo quadrato
	display.hline(x + space + size + 10, y, size, colorb)
	display.vline(x + space + size + 10, y, size, colorb)
	display.vline(x + space + (size) * 2, y, size, colorb)
	display.hline(x + space + size + 10, y + size, size, colorb)

	# Terzo quadrato
	display.hline(x + space + (size + 10) * 2, y, size, colorc)
	display.vline(x + space + (size + 10) * 2, y, size, colorc)
	display.vline(x + space + (size + 9) * 3, y, size, colorc)
	display.hline(x + space + (size + 10) * 2, y + size, size, colorc)

def touch_listener(button):
	#change_text(button)
	uasyncio.run(touch_listener_async(button))

async def touch_listener_async(button):
	global inapp
	pressed = touch.state(button)
	uasyncio.sleep_ms(5000)
	pressed = touch.state(button)

	if pressed and button == "A":
		inapp = lastapp
		print("nav",inapp)

	if pressed and button == "B":
		inapp = ""
		print("nav","back")

def persistent():
	app1 = app[0]
	app2 = app[1]
	app3 = app[2]

	colorapp1 = 0xffffff
	colorapp2 = 0xffffff
	colorapp3 = 0xffffff

	if idx > 2:
		app1 = app[idx-2]
		app1 = app[idx-1]
		app1 = app[idx]
	if app1 == inapp:
		colorapp1 = 0xf8db5E
	if app2 == inapp:
		colorapp2 = 0xf8db5E
	if app3 == inapp:
		colorapp3 = 0xf8db5E

	display.text(app1,50,175,colorapp1)
	display.text(app2,250,175,colorapp2)
	display.text(app3,450,175,colorapp3)
	
	display.show()

#uasyncio.run(battery_level())
#touch.callback(touch.BOTH, touch_listener)

#display.text("MONOCLE",250,175,0xffffff)
#display.show()
```

josuah, 2023-03-04:
Further feedback on this: it looks like using different way to load the code leads to different results:

> the same code loaded from the desktop or from the phone does not go the same way, load the same first part  then crashes on the first touch

It could be worth comparing raw mode with (friendly) paste mode.

siliconwitch, 2023-03-24:
Doesn't seem to be a touch issue, but rather a hardfault when `display.show()` is called. Try the code below:

Uncommenting the display.show() will result in hardfault

```python
import display
import device
import uasyncio
import gc
import time
import touch

battlev = ""
firstaccess = True
battlevint = 100
app = ["phone", "camera" , "next?"]
idx = 0
alive = "MONOCLE"
inapp = ""
time = "x"

async def battery_level():
	global battlevint
	
	if device.battery_level() < battlevint:
		battlevint = device.battery_level()

	battlev = str(battlevint) + '%'
	display.text(battlev, 0,0,0xffffff)
	uasyncio.sleep_ms(5000)

def change_text(button):
	global app
	global idx
	global alive
	global lastapp
	global nextapp

	lastapp = alive
	limit = len(app) - 1
	
	if button == "A":
		idx = idx + 1
		if idx > limit:
			idx = 0
	if button == "B":
		idx = idx - 1
		if idx < 0:
			idx = limit
	alive = app[idx]

	draw_ui()
	persistent()
	print(alive)

def draw_ui():
	
	global idx
	uasyncio.run(battery_level())

	x = 1
	y = 100
	size = 200
	colora = 0xffffff
	colorb = 0xffffff
	colorc = 0xffffff
	space = 10

	if idx <= 2:
		if idx == 0:
			colora = 0xf8db5E
		if idx == 1:
			colorb = 0xf8db5E
		if idx == 2:
			colorc = 0xf8db5E

	# Primo quadrato
	display.hline(x, y, size, colora)
	display.vline(x, y, size, colora)
	display.vline(x + size, y, size, colora)
	display.hline(x, y + size, size, colora)

	# Secondo quadrato
	display.hline(x + space + size + 10, y, size, colorb)
	display.vline(x + space + size + 10, y, size, colorb)
	display.vline(x + space + (size) * 2, y, size, colorb)
	display.hline(x + space + size + 10, y + size, size, colorb)

	# Terzo quadrato
	display.hline(x + space + (size + 10) * 2, y, size, colorc)
	display.vline(x + space + (size + 10) * 2, y, size, colorc)
	display.vline(x + space + (size + 9) * 3, y, size, colorc)
	display.hline(x + space + (size + 10) * 2, y + size, size, colorc)

async def touch_listener_async(button):
	global inapp
	pressed = touch.state(button)
	uasyncio.sleep_ms(5000)
	pressed = touch.state(button)

	if pressed and button == "A":
		inapp = lastapp
		print("nav",inapp)

	if pressed and button == "B":
		inapp = ""
		print("nav","back")

def touch_listener(button):
	change_text(button)
	uasyncio.run(touch_listener_async(button))

def persistent():
	app1 = app[0]
	app2 = app[1]
	app3 = app[2]

	colorapp1 = 0xffffff
	colorapp2 = 0xffffff
	colorapp3 = 0xffffff

	if idx > 2:
		app1 = app[idx-2]
		app1 = app[idx-1]
		app1 = app[idx]
	if app1 == inapp:
		colorapp1 = 0xf8db5E
	if app2 == inapp:
		colorapp2 = 0xf8db5E
	if app3 == inapp:
		colorapp3 = 0xf8db5E

	display.text(app1,50,175,colorapp1)
	display.text(app2,250,175,colorapp2)
	display.text(app3,450,175,colorapp3)
	
	# display.show()

uasyncio.run(battery_level())
touch.callback(touch.BOTH, touch_listener)

display.text("MONOCLE",250,175,0xffffff)
# display.show()
```

siliconwitch, 2023-03-24:
Closing as we're about to depreciate the old display library anyway

closed 2023-03-24 by siliconwitch