brilliantlabsAR/monocle-micropython #146: [bug] vline/hline not drawn near the max x/y

josuah, 2023-03-11:
https://discord.com/channels/963222352534048818/1074358904168923146/1084204293084688424

@jcun4128
> `vline(0, 0, 400, ...)` does not appear on left corner as expected
> I would expect this to draw a box around the display perimeter
>
> ```
> import display
>
> display.vline(0, 0, 400, 0xffffff)
> display.hline(0, 0, 640, 0xffffff)
> display.text("Controls", 260, 100, 0xffffff)
> display.hline(0, 392, 640, 0xffffff)
> display.vline(632, 0, 400, 0xffffff)
>
> display.show()
> ```
>
> left one is missing

Thank you for reporting!
We can then test that with the next library.

jdc-cunningham, 2023-03-19:
Adding this example

```
import display

display.line(50, 50, 150, 150, 0xffffff)

display.show()
```

Does not draw a line

Also the varying thickness of lines between hline/vline and line, would expect them to be the same (default no thickness specified)

siliconwitch, 2023-03-20:
Closing as a big display update is coming very soon in the new FPGA binary

closed 2023-03-20 by siliconwitch