brilliantlabsAR/monocle-micropython #148: [feature] free all memory associated with an "app"

josuah, 2023-03-13:
Allow set of functions be declared and withdrawn all at once, together building an application.

This would allow running multiple "applications" that would otherwise not fit in RAM, by swapping them one after the other.

siliconwitch, 2023-03-20:
This can be done by using the `del()` operator, followed by `gc.collect()`

josuah, 2023-04-14:
Something that could come of help with that:
https://discordapp.com/channels/574275045187125269/574275045611012097/1011271986351198310

> I need to think about it more (totally untested code to follow), but I think that's actually do-able
>
> ```python
> class temp_scope:
>   def __enter__(self):
>     self.k = set(sys.modules.keys())
>
>   def __exit__(self, ...):
>     for k in set(sys.modules.keys()) - self.k:
>       del sys.modules[k]
>
> with temp_scope():
>     from foo import *
>     print(bar(42))
> ```
>
> If `temp_scope()` took `globals()` as an arg, you could do the same to delete any new keys on globals
> but it would handle freeing any nested modules (applying to both scenarios that kevin's post describes)

closed 2023-05-04 by siliconwitch