brilliantlabsAR/monocle-micropython #151: New `display` library

siliconwitch, 2023-03-20:


josuah, 2023-03-30:
Not sure how to use the FBGraphics block yet. This one has the SPI API for sending graphical objects.
See https://github.com/brilliantlabsAR/monocle-fpga/issues/9

josuah, 2023-03-30:
https://github.com/brilliantlabsAR/monocle-micropython/pull/159/files

josuah, 2023-04-03:
Currently blocked by the bitstream transfer time on every startup.

josuah, 2023-04-03:
Currently blocked by the choice for a new API for the display module...
Or mabye nothing at first, letting user decide?

josuah, 2023-04-07:
- [ ] document the API
- [ ] implement the API as shown below
- [ ] flip the display vertically and horizontally (180° spin)
- [ ] check the coordinate system for both text so that (0, 0) is at the bottom left

API:

```python
display.Line()
display.Rectangle()
display.Polygon()
display.Polyline()
display.Text()

.move()

display.show()
```

Usage:

```python
l1 = display.Line()
display.show(l1)

l2 = display.Line()
p1 = display.Polygon()
grp = [l1, l2, p1]
display.show(grp)

l1.move()
grp.move()
```

josuah, 2023-04-13:
Fixed by #173 (WIP)

closed 2023-05-01 by siliconwitch