brilliantlabsAR/monocle-micropython #152: [feature] wake from bluetooth

josuah, 2023-03-20:
Queried from the chat.

https://discord.com/channels/963222352534048818/984966420603482182/1087157969596198993

josuah, 2023-03-20:
ECS336CN is the last part that consumes a lot of power, it could be turned off.

siliconwitch, 2023-03-24:
Similar how the camera sleeping is automatically handled within the driver, perhaps `display.clear()` could shut down the screen, and `display.show()` turns it back on again. Then the power stuff is transparent for the user and quite obvious that when the display is blank, it consumes the least amount of energy.

siliconwitch, 2023-04-07:
This is now done using `fpga.run()`. If you do `fpga.run(False)`, it's the lowest power mode because it shuts down all the other rails except for the nRF52

closed 2023-04-07 by siliconwitch