brilliantlabsAR/monocle-micropython #162: [feature] Add `input()`, maybe more?

josuah, 2023-04-03:
https://discord.com/channels/963222352534048818/1092348878868709376

Since we have plenty of flash room, and yet an extra external flash, does it make sense to turn on more features?

`micropython/py/mpconfig.h`

```
#define MICROPY_PY_BUILTINS_INPUT (MICROPY_CONFIG_ROM_LEVEL_AT_LEAST_EXTRA_FEATURES)
```

We can therefore enable either `MICROPY_PY_BUILTINS_INPUT` individually, or switch to the default level `MICROPY_CONFIG_ROM_LEVEL_AT_LEAST_EXTRA_FEATURES`.

No preference for me.

josuah, 2023-04-03:
Related missing feature is `sys.stdin`:

`micropython/py/mpconfig.h`
```
#define MICROPY_PY_SYS_STDFILES (MICROPY_CONFIG_ROM_LEVEL_AT_LEAST_EXTRA_FEATURES)
```

josuah, 2023-04-03:
Temporary workardoun:

```python
def coroutine():
    while True:
        input = yield
        print(f'input="{input}"')

co = coroutine()
next(co)
co.send("this is what the user typed")
```

siliconwitch, 2023-04-03:
> switch to the default level MICROPY_CONFIG_ROM_LEVEL_AT_LEAST_EXTRA_FEATURES.

Great idea!

I'm in two minds if the external flash will be fast enough for loading files, so maybe we'll need to enable internal flash storage too, but for now, we can definitely enable the feature on its own. If later on, we don't see problems with the external flash speed, then we can enable all the MicroPython features for sure.

commit 0ca1b121 (2023-04-03): turn on input() support and implement support for it

fixes #162
PR #164 (2023-04-04): Turn on and implement input()
This fixes #162