brilliantlabsAR/monocle-micropython #171: [bug] only 500 free bytes reported for the flash

josuah, 2023-04-12:
This code reported 500 on a system that is not yet filled-up:

```python
f_bsize   = 0  # file system block size
f_frsize  = 1  # fragment size
f_blocks  = 2  # size of fs in f_frsize units
f_bfree   = 3  # number of free blocks
f_bavail  = 4  # number of free blocks for unprivileged users
f_files   = 5  # number of inodes
f_ffree   = 6  # number of free inodes
f_favail  = 7  # number of free inodes for unprivileged users
f_flag    = 8  # mount flags
f_namemax = 9  # maximum filename length

sv = os.statvfs("/run")
print(sv)

free = sv[f_bfree] * sv[f_bsize]
total = sv[f_blocks] * sv[f_bsize]
print(f"free file storage: {round(free / total * 100)}%")
```

https://discord.com/channels/963222352534048818/1074359320969490452/1094942765470732329

josuah, 2023-04-18:
While the watchdog timer is a precious help to revive the Monocle if anything gets stuck, it plays us trick on slow operations like writing to flash:

This is the code known to produce an OS Error 28:

```python
import gc

f = open('one.py', 'w')
f.write('def text(display_ref):\n')
f.write('  display_ref.text("one", 0, 0, 0xffffff)\n')
f.close()

gc.collect()

f = open('two.py', 'w')
f.write('def text(display_ref):\n')
f.write('  display_ref.text("two", 0, 0, 0xffffff)\n')
f.close()

gc.collect()

f = open('main.py', 'w')
f.write('import one\n')
f.write('import two\n')
f.write('import time\n')
f.write('import display as display_ref\n')

gc.collect()

f.write('display_ref("main", 0, 0, 0xffffff)\n')
f.write('time.sleep(3)\n')
f.write('one.text(display_ref)\n')
f.write('time.sleep(2)\n')

gc.collect()

f.write('two.text(display_ref)\n')
f.close()
```

This is what happened after some delay:

```
Program received signal SIGTRAP, Trace/breakpoint trap.
0x00036952 in WDT_IRQHandler ()
(gdb) bt
#0  0x00036952 in WDT_IRQHandler ()
#1  <signal handler called>
#2  0x0004a352 in qstr_data ()
#3  0x0004a530 in mp_vprintf ()
#4  0x0004a5b0 in mp_printf ()
#5  0x00042e26 in mp_obj_print_exception ()
#6  0x00035566 in parse_compile_execute.lto_priv ()
#7  0x0003b44a in main ()
```

So it might be that some operation was too slow and triggered the watchdog.

This was tried on `667bbc1` and will now try again on `main`.

siliconwitch, 2023-04-18:
I don't think any watchdog is currently enabled right?

josuah, 2023-04-18:
Good point! That escaped my mind...

It could be due to **W**DT being the last in alphabetical order, and all handlers annealed to the same name have that "WDT" name...

The RTT logs are more insightful:
```
MicroPython on Monocle - v23.108.0856 (67fac4ebc)
Booting FPGA from internal flash
Camera not detected
Softdevice using 0x2A60 bytes of RAM
[nrf52.cpu] halted due to breakpoint, current mode: Handler HardFault
xPSR: 0x810b0003 pc: 0x00036952 msp: 0x2000fcd0
App error: 0xDEADBEEF at monocle-core/monocle-startup.c:40
```

siliconwitch, 2023-04-18:
Yeah, that's probably a hard fault. Dereferencing null pointer or something like that?

josuah, 2023-04-18:
By adding a bit of logging to this function like so:

```python
mp_obj_t mp_stream_write(mp_obj_t self_in, const void *buf, size_t len, byte flags) {
    NRFX_LOG(".---. %s", __func__);
    int error;
    mp_uint_t out_sz = mp_stream_rw(self_in, (void *)buf, len, &error, flags);
    if (error != 0) {
        if (mp_is_nonblocking_error(error)) {
            // http://docs.python.org/3/library/io.html#io.RawIOBase.write
            // "None is returned if the raw stream is set not to block and
            // no single byte could be readily written to it."
            return mp_const_none;
        }
        NRFX_LOG("'---' %s !!", __func__);
        mp_raise_OSError(error);
    } else {
        NRFX_LOG("'---' %s %d", __func__, out_sz);
        return MP_OBJ_NEW_SMALL_INT(out_sz);
    }
}
```
We get:
```
...
.---. mp_stream_write
'---' mp_stream_write 2
.---. mp_stream_write
'---' mp_stream_write 1
.---. mp_stream_write
'---' mp_stream_write 11
.---. mp_stream_write
'---' mp_stream_write 2
.---. mp_stream_write
.---. mp_stream_write
[nrf52.cpu] halted due to breakpoint, current mode: Handler HardFault
xPSR: 0x810f0003 pc: 0x000369aa msp: 0x2000fc10
'---' mp_stream_write !!
App error: 0xDEADBEEF at monocle-core/monocle-startup.c:40
```

Which means there was a recursive call of `mp_stream_write` followed by an error on the nested write, followed by possibly a mishandling of that error resulting in the crash we see.

josuah, 2023-04-25:
Some more GDB crash backtrace:
```
0x0003624a in WDT_IRQHandler ()
Source directories searched: /home/josuah/brilliantlabs/monocle-micropython:$cdir:$cwd
(gdb) bt
#0  0x0003624a in WDT_IRQHandler ()
#1  <signal handler called>
#2  0x00027b62 in __sfputs_r (ptr=0x20002a64 <impure_data>, fp=0x20003c54, buf=<optimized out>,
    len=<optimized out>) at ../../../../../../../../newlib/libc/stdio/nano-vfprintf.c:428
#3  0x000280c6 in _printf_i (data=data@entry=0x20002a64 <impure_data>, pdata=pdata@entry=0x2000fa58,
    fp=fp@entry=0x20003c54, pfunc=pfunc@entry=0x27b41 <__sfputs_r>, ap=<optimized out>,
    ap@entry=0x2000fa54) at ../../../../../../../../newlib/libc/stdio/nano-vfprintf_i.c:241
#4  0x00027dea in _vfprintf_r (data=data@entry=0x20002a64 <impure_data>, fp=0x20003c54,
    fmt0=fmt0@entry=0x51a8b "", ap=..., ap@entry=...)
    at ../../../../../../../../newlib/libc/stdio/nano-vfprintf.c:636
#5  0x000265e0 in printf (fmt=0x4c7cf "%s:%d:error: No more free space %lu%s\n")
    at ../../../../../../../../newlib/libc/stdio/printf.c:56
#6  0x0002d416 in lfs2_alloc ()
#7  0x0002d576 in lfs2_file_relocate ()
#8  0x0002d7f2 in lfs2_file_flushedwrite.lto_priv ()
#9  0x00032a50 in mp_vfs_lfs2_file_write ()
#10 0x0003d74c in mp_stream_rw ()
#11 0x0002b25a in mp_stream_write.constprop ()
#12 0x0003c0c0 in stream_write_method.lto_priv ()
#13 0x00042718 in fun_builtin_var_call.lto_priv ()
#14 0x0003b742 in mp_execute_bytecode ()
#15 0x00041180 in fun_bc_call ()
#16 0x00034f18 in parse_compile_execute.lto_priv ()
#17 0x0003a7d4 in main ()
(gdb)
```

Interesting clue: "no more free space" in the error string...

josuah, 2023-04-25:
This might be our cause: `LFS2_ERR_CORRUPT`:

```C
3379         // program as much as we can in current block
3380         lfs2_size_t diff = lfs2_min(nsize, lfs2->cfg->block_size - file->off);
3381         while (true) {
3382             int err = lfs2_bd_prog(lfs2, &file->cache, &lfs2->rcache, true,
3383                     file->block, file->off, data, diff);
3384             if (err) {
3385                 if (err == LFS2_ERR_CORRUPT) {
3386                     goto relocate;                <<< We might have gotten here
3387                 }
3388                 file->flags |= LFS2_F_ERRED;
3389                 return err;
3390             }
3391
3392             break;
3393 relocate:
3394             err = lfs2_file_relocate(lfs2, file);      <<< Which leads us here
3395             if (err) {                                     visible in the backtrace
3396                 file->flags |= LFS2_F_ERRED;
3397                 return err;
3398             }
3399         }
```

josuah, 2023-04-25:
What happens: LFS2 keeps relocating the block over and over and over, until it crossed every block and could not find a single one which is valid:

`one.py` works:
```
f = open('one.py', 'w')
f.write('def text(display_ref):\n')
f.write('  display_ref.text("one", 0, 0, 0xffffff)\n')
f.close()

.---. mp_stream_write
'---' mp_stream_write 23
.---. mp_stream_write
'---' mp_stream_write 2
.---. mp_stream_write
'---' mp_stream_write 1
.---. mp_stream_write
'---' mp_stream_write 42
.---. mp_stream_write
'---' mp_stream_write 2
.---. mp_stream_write
'---' mp_stream_write 1
MP_BLOCKDEV_IOCTL_BLOCK_ERASE(0)
MP_BLOCKDEV_IOCTL_SYNC(0)
```

`two.py` works
```
f = open('two.py', 'w')
f.write('def text(display_ref):\n')
f.write('  display_ref.text("two", 0, 0, 0xffffff)\n')
f.close()

.---. mp_stream_write
'---' mp_stream_write 23
.---. mp_stream_write
'---' mp_stream_write 2
.---. mp_stream_write
'---' mp_stream_write 1
.---. mp_stream_write
'---' mp_stream_write 42
.---. mp_stream_write
'---' mp_stream_write 2
.---. mp_stream_write
'---' mp_stream_write 1
MP_BLOCKDEV_IOCTL_SYNC(0)
```

`main.py` is longer, and does not work... See how it tries to erase every single block, and halts once it looped through everything.
```
f = open('main.py', 'w')
f.write('import one\n')
f.write('import two\n')
f.write('import time\n')
f.write('import display as display_ref\n')
f.write('display_ref("main", 0, 0, 0xffffff)\n')
f.write('time.sleep(3)\n')
f.write('one.text(display_ref)\n')
f.write('time.sleep(2)\n')
f.write('two.text(display_ref)\n')
f.close()

MP_BLOCKDEV_IOCTL_BLOCK_COUNT(0)
.---. mp_stream_write
'---' mp_stream_write 11
.---. mp_stream_write
'---' mp_stream_write 2
.---. mp_stream_write
'---' mp_stream_write 1
.---. mp_stream_write
'---' mp_stream_write 11
[... 154 lines cut ...]
MP_BLOCKDEV_IOCTL_BLOCK_ERASE(41)
MP_BLOCKDEV_IOCTL_BLOCK_ERASE(42)
MP_BLOCKDEV_IOCTL_BLOCK_ERASE(43)
MP_BLOCKDEV_IOCTL_BLOCK_ERASE(44)
MP_BLOCKDEV_IOCTL_BLOCK_ERASE(45)
MP_BLOCKDEV_IOCTL_BLOCK_ERASE(46)
MP_BLOCKDEV_IOCTL_BLOCK_ERASE(47)
MP_BLOCKDEV_IOCTL_BLOCK_ERASE(48)
MP_BLOCKDEV_IOCTL_BLOCK_ERASE(49)
MP_BLOCKDEV_IOCTL_BLOCK_ERASE(50)
MP_BLOCKDEV_IOCTL_BLOCK_ERASE(51)
MP_BLOCKDEV_IOCTL_BLOCK_ERASE(52)
MP_BLOCKDEV_IOCTL_BLOCK_ERASE(53)
MP_BLOCKDEV_IOCTL_BLOCK_ERASE(54)
MP_BLOCKDEV_IOCTL_BLOCK_ERASE(55)
MP_BLOCKDEV_IOCTL_BLOCK_ERASE(56)
MP_BLOCKDEV_IOCTL_BLOCK_ERASE(57)
MP_BLOCKDEV_IOCTL_BLOCK_ERASE(58)
MP_BLOCKDEV_IOCTL_BLOCK_ERASE(59)
[nrf52.cpu] halted due to breakpoint, current mode: Handler HardFault
xPSR: 0x810e0003 pc: 0x000362ba msp: 0x2000fc08
MP_BLOCKDEV_IOCTL_BLOCK_ERASE(60)
'---' mp_stream_write !!
App error: 0xDEADBEEF at monocle-core/monocle-startup.c:40
```

josuah, 2023-04-25:
Therefore, the problem appears to be: whenever a file is larger than a single block, the second blocks it spans across always appears to be corrupted.

Quickfix: increase the size of the blocks, to give us more time to work on this.

josuah, 2023-04-26:
Root cause: #184

closed 2023-04-26 by josuah