brilliantlabsAR/monocle-micropython #182: `monocle.js` to allow any page writing to the Monocle

josuah, 2023-04-24:
Suggested on Discord:
https://discord.com/channels/963222352534048818/1074359320969490452/1100052037242208308

It would need some filtering to prevent advertising to be displayed from any website.
For now the web browser connection prompt acts as a filter already: needs user acknowledgement.

Utopiah, 2023-04-24:
See https://github.com/semtexzv/monoapp/blob/master/src/lib/monocle.ts with a usage example 

```javascript
const cmd: string = `import display; display.text("${$total}", 0, 0, 0xFFFFFF); display.show()\r`;
(monocle as MonocleData).repl(cmd)
```

Utopiah, 2023-04-24:
Alternatively built on `repl.js` https://github.com/jdc-cunningham/bl-monocle-reactjs-pwa/blob/poc-bluetooth-only/app/src/App.js#L21

Utopiah, 2023-04-24:
This could help make the documentation interactive and could facilitate demonstration or workshop materials on e.g Glitch or CodePen to be easily remixable with whatever environment or framework used.

Utopiah, 2023-04-25:
Did https://monoclejs.glitch.me/

siliconwitch, 2023-06-16:
Closing for now as this is less firmware related

closed 2023-06-16 by siliconwitch