brilliantlabsAR/monocle-micropython #187: Display API: memory allocation failed

endanke, 2023-05-02:
I'm trying to use the display API using the latest release (`v23.112.1046`) with the following two lines:
`import display; line = display.Line(500, 5, 500, 395, display.WHITE); display.show(line);`
`import display; line = display.Line(520, 5, 520, 395, display.WHITE); display.show(line);`

The first one works, but the second one produces this error:
```
Traceback (most recent call last):
  File "<stdin>", in <module>
  File "display.py", in show
MemoryError: memory allocation failed, allocating 16384 bytes
```

Is this expected or a bug?

siliconwitch, 2023-05-05:
Is this the same as https://github.com/sathibault/vgrs/issues/8 @josuah ?

josuah, 2023-05-05:
It is likely the same issue, and the latest dev branch from `vgrs` fixes it!
[image]

endanke, 2023-05-06:
@josuah @siliconwitch I can confirm that it works now, thanks a lot! :) Now I can render lines anywhere on the screen. :D 🎉

closed 2023-05-06 by endanke