brilliantlabsAR/monocle-micropython #198: Text output randomly mixed-up

josuah, 2023-05-11:
Upon certain conditions, as well as randomly with the same code, the text output can become offset.

@jdc-cunningham did this reproducer:
https://gist.github.com/jdc-cunningham/ff37424802a8db87c317c7d7c5575f00 - breaking
https://gist.github.com/jdc-cunningham/117a4e38177043c03631d711db94691b - works

And developed the problem here: https://discord.com/channels/963222352534048818/1105933108617945268/1105954663204921416

Here the 46m should be at the top right corner of the display:
[image]

siliconwitch, 2023-05-22:
Is this fixed now with the latest release?

josuah, 2023-05-24:
Some of it is fixed, and the remaining bug will be discussed in #208.
Thank you!

closed 2023-05-24 by josuah