brilliantlabsAR/monocle-micropython #204: `fbtext` shows text with an offset

josuah, 2023-05-16:
The long-term fix is to implement a better collision detection that never make overlapping text be passed to the FPGA.

A short-term workaround would be to allow to reset the fbtext module every time text is sent, or eventually manually by the developer.

The problem is once overlapping text occurs, the whole display is completely offset even if the next commands are valid.

Because the offset also happens randomly, it means a single occurrence of this bugs requires a full device reboot to be corrected.

siliconwitch, 2023-05-22:
Related to #198 ?

josuah, 2023-05-24:
This is indeed, and it is further discussed in #208 which is related to a different bug with the same effect.
Thank you!

closed 2023-05-24 by josuah