brilliantlabsAR/monocle-micropython #267: Bluetooth: allow to change the bluetooth name

josuah, 2023-09-27:
This was requested by an user who wanted to adjust the name of the Bluetooth interface.

siliconwitch, 2023-10-20:
Bluetooth central devices wouldn't like this. They often cache the name so random name changes would create weird behavior. Usually, the fix is to reboot the Bluetooth central, but unless the users know about this, it might cause frustration because they might not be able to connect or scan properly etc

louisbarrett, 2023-10-20:
> Bluetooth central devices wouldn't like this. They often cache the name so random name changes would create weird behavior. Usually, the fix is to reboot the Bluetooth central, but unless the users [...]

The assumption on my part is that most people aren't connecting to their devices by name because that could be problematic from the start due to duplicate names, spoofing etc. 

I think this is a reasonable ask since the monocle is being discovered either by scanning for the GUID on characteristics (immutable), or the MAC address (also immutable). 

Also for security reasons its a good idea to allow users to do this. 

P.S - Are there any settings on the monocle that accept persistent changes today?

josuah, 2023-10-23:
> P.S - Are there any settings on the monocle that accept persistent changes today?

That  would need to be a custom build:
https://github.com/brilliantlabsAR/monocle-micropython/blob/main/main.c#L742